mylonics/zephyr-ide #109: Cannot run application when running on Devcontainer

J-Azevedo, 2025-04-21:
I am currently trying to use this plugin in my development, I have Devcontainer that setups an workspace where Zephyr is installed and configured, with the following structure:

workspace
|___my-app
|______app
|_________src
|______docs
|______submanifest
|______west.yml
|___zephyr

The app is already configured and running, and I able to build it using `west build`.

However, when I try to run it using the Zephyr IDE, I cannot create a build configuration

[image]

I choose the Zephyr Directory Only option, and it gives me always the error 

[image]

But, the Error log doesn't appear in any place,  so I cannot understand what might be the issue. How can I have more information about this error? And what might be happening

**Desktop (please complete the following information):**
 - OS: [Ubuntu Docker Container inside WSL ]
 - vscode Version: 1.99.3
 - Zephyr IDE Version:  1.7.0
 - Zephyr Version: 4.1
 - Zephyr SDK Version: 0.27.5
 - Terminal: bash

rijesha, 2025-04-21:
There should be an output panel, and you can select Zephyr IDE. Hopefully there is a log message there. 

https://code.visualstudio.com/docs/getstarted/userinterface#_basic-layout (panel label D has an output tab)

rijesha, 2025-04-22:
Could you try running west update and see if it works then.

If that does work then it is just a workspace configuration that isn't saved when using a workspace that has already been set up. I think a fix is very doable.

Do you mind sharing the dev container with setup instructions with me?

J-Azevedo, 2025-04-22:
I ran the west update, but it will always try to run in my application path, not in the Zephyr path

[image]

I am selecting workspace as the Zephyr Setup Location

[image]

If i try to Select the Folder to be specifically pointing to /workspace/zephyr. The Zephyr IDE doesn't seem able to find the correct path

[image]

rijesha, 2025-04-22:
Could you share the dev container with me please?
Did you create the dev container starting with a zephyr-ide repository or is this a preexisting dev container you now want to use with zephyr-ide?

What is in your west.yml file?

closed 2025-08-11 by rijesha