mylonics/zephyr-ide #111: Specify location of a runner, if not found in PATH

Gerriko, 2025-05-20:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

I am offered the option to add a runner, which then provides me with a number of options. The IDE then assumes that these runners are available in the PATH.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

I would like an option where I can specify the exact location of the runner as in my case the runner is not in the PATH but in a folder inside my workspace. For some reason it cannot seem to find the exectable  (in my case it is bossac).

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

Gerriko, 2025-05-20:
I did observe that when you select a specific runner it offers for you the option to set arguments. It's not clear if you can enter a path here or not.

Gerriko, 2025-05-27:
> Is there a way of doing this through west? i.e is there an argument to west flash that you can pass that will specify the path of the runner?

yes. You simply use the path to where your bossac application resides in the command line (can be relative or actual path)
e.g. west flash --runner path/to/bossac

rijesha, 2025-06-11:
Here is the code for adding arguments to a runner.

```
  let cmd = `west flash --build-dir ${path.join(wsConfig.rootPath, project.rel_path, build.name)}`;

  if (runner.runner !== "default") {
    cmd += ` -r ${runner.runner}`;
  }
  cmd += ` ${runner.args}`;
```

So I think it should just work. What happens when you add your argument?

Gerriko, 2025-06-14:
I'm away for bit. Will try this in a week or so.

rijesha, 2025-08-01:
Closing as stale