mylonics/zephyr-ide #113: Option to select board revisions

dolence, 2025-06-12:
**Is your feature request related to a problem? Please describe.**
I have a custom board with two revisions: stm32h7_core@original (default if not specified) and stm32h7_core@lvps. When adding a build, I can't select which board revision it will be using, even if I try to manually specify it. The lack of this feature make this otherwise very nice extension unusable to me.

[image]

**Describe the solution you'd like**
when available, board revisions should be listed like this: 
```
boardname@default_revision
boardname@rev_1
...
boardname@rev_n
```

rijesha, 2025-06-13:
I have issued an update for this. It is version 1.7.5.

The revisions work for a zephyr version greater than v4.1.0 due to implementing this feature using a feature that is currently only available in main (https://github.com/zephyrproject-rtos/zephyr/commit/01b0dec496a7cd3101e6004f00ff0ebff2cda894)

If you are using the apps minimal west configs, you will need to update cmsis to cmisis6 if changing the zephyr version to main. 

If you would like to use revisions on v4.1.0 then I would recommend modifying the zephyr-ide.json board field under the build configuration.

rijesha, 2025-06-22:
Thanks I appreciate the feedback.

I'm going to close this issue for now. If in the future someone needs this support for the 3.7 LTS, I will go back and see what we can do.

closed 2025-06-22 by rijesha