mylonics/zephyr-ide #114: VSCode on Mac requires changes to settings.jsion

i2pi, 2025-06-12:
When initializing a new workspace on MacOS, I get and error message from pip citing a PEP668 violation (trying to install globally, not in the virtual env).

I saw the note in the manual about installing the Python Virtual Environments extension, but that didn't help. In the end I found that adding these two lines to my settings.json fixed it:

```
    "terminal.integrated.env.osx": {
     "PATH": "${workspaceFolder}/.venv/bin:/opt/homebrew/bin/:/Applications/STMicroelectronics/STM32Cube/STM32CubeProgrammer/STM32CubeProgrammer.app/Contents/MacOs/bin/:${env:PATH}"
    },
    "terminal.integrated.automationProfile.osx": {
     "path": "/bin/zsh",
     "args": ["-i"]
   }
```

rijesha, 2025-06-13:
Thanks for the issue. I will take note of it and keep this issue open so others can see your work around.

Modifying the setting.json like this is not the ideal solution. I have been having issues with modifying the path environment in zsh and MacOS. It is something inherent with how vscode handles it on that platform.

I am hoping to get access to some mac hardware in the future for better integration.

rijesha, 2025-08-01:
Issue seems to be fixed with release v1.7.7.