mylonics/zephyr-ide #120: Add support for Cursor?

anthonyhoegberg, 2025-08-08:
Is there any way this extension could be configured/setup to work with cursor?

rijesha, 2025-08-08:
I decided to see what copilot can do #122.

anthonyhoegberg, 2025-08-12:
> Ok I have added the extension to the Open VSX registry.
> 
> The extension can run by itself, but also has integration with dts-lsp and "C_Cpp.default.compileCommands": "${workspaceFolder}\.vscode\compile_commands.json"
> 
> What opensource extension do you recommend for cpp intellisense and do you know if it has a settings variable or api to set a compile_commands.json.

Not quite sure about this one, maybe clang? i remember reading smth about this in the platformio repository bc they are facing similar issues with cpp intellisense etc, i also think cursor itself are releasing open source versions of c++ extension

rijesha, 2025-08-12:
dts-lsp is a standalone that isn't relying on anything. It looks like the extension should already be compatible with cursor.

dts-lsp provides intellisense for .dts files and that exists in open vsx. clangd seems like the appropriate choice for cursor intellisense, and it should already be working, based on how it detects for compileCommands.

So theoretically the extension should be compatible now. If there is a specific cpp extension you would like zephyr-ide to support in the future, that can be a new issue. If clangd isn't working for intellisense then let me know as well, there seems to be a fallback parameter that can be set if it can't find a compileCommands file.

rijesha, 2025-08-13:
I shall close this issue as the main scope was getting the extension into open vsx.
If you find any compatibility issues or if you would like to go ahead with the server so that an AI can interact directly with the extension we can make that a new issue and go from there.