mylonics/zephyr-ide #182: Setup panel proposes to install WIndws tools into WSL2/ubuntu

Josverl, 2025-08-27:
Before creating an issue have you checked for past issues that may be relevant, checked [manual.md](https://github.com/mylonics/zephyr-ide/blob/main/docs/MANUAL.md) for any reference to your issue, and followed any additional setup instructions specific to your board/architecture as specified in the zephyr docs?

**Describe the bug**
A clear and concise description of what the bug is.
- open WLS2 / ubuntu terminal 
- open a floder in vscode ( ` code ~/micropython`  ) 
- Zephyr IDE > Open setup Panel
- Host tools shows Linux - but proposed to install tooling via `winget`  

Possibly the extension does not deal correctly with remote development or mixes up the local host OS with the remote host OS.
https://code.visualstudio.com/docs/remote/remote-overview

In my case: 
 - Host OS : Windows 11 
 - Remote OS : WSL2/Ubuntu 22.04.5 LTS

**Desktop (please complete the following information):**
Version: 1.103.2 (user setup)
Commit: 6f17636121051a53c88d3e605c491d22af2ba755
Date: 2025-08-20T16:45:34.255Z
Electron: 37.2.3
ElectronBuildId: 12035395
Chromium: 138.0.7204.100
Node.js: 22.17.0
V8: 13.8.500258-electron.0
OS: Windows_NT x64 10.0.26100 ( Win11) 
 - Zephyr IDE Version: 2.0.2
 - Zephyr Version:  4.2 already installed 
 - Zephyr SDK Version:  already installed 
 - Terminal: n/a - bash

rijesha, 2025-09-08:
Thanks for reporting the issue. I will have a look into when I have some time.

rijesha, 2025-10-31:
I have a potential fix in pre-release along with a few other host tool install changes. I will be doing some testing of it. We can reopen this issue if the problem persists.