mylonics/zephyr-ide #189: Avoid replacing env-var ZEPHYR_SDK_INSTALL_DIR, etc.

flisboac, 2025-09-15:
Would it be possible to avoid replacing the `ZEPHYR_SDK_INSTALL_DIR`, `ZEPHYR_BASE`, etc., environment variables, if they're already configured for the environment?

I have a VSCode Workspace template that's based around [Devcontainers](https://containers.dev/). I do all my work inside a Docker container, which is built based on the needs of the Zephyr project I'm working on ATM, and therefore already has all the necessary tools installed at system or West-venv level. It has exactly the versions needed for SDK and tools, and whatnot.

In other words, in some cases, there's no need for another tool to assume control of those Zephyr-related env-vars. It would be nice to have the option to disable manipulation of env-vars, or at least give the option to force a specific value for some of them in `settings.json`.

In my case, specifically:

- My Zephyr app declares dependency to a specific Zephyr version in its `west.yml`.
- `ZEPHYR_BASE` is configured to be the Zephyr cloned by `west init`. This allows me to open (in VSCode, etc.) and use the exact Zephyr version my application needs;
- `ZEPHYR_SDK_INSTALL_DIR` points to a Zephyr SDK installed as part of the Docker image's build process. Its version is fixed in the Dockerfile.

In a way, my environment is prepared to automate as much of the boring stuff as possible. The intention is for any developer to just clone, run a command or two and open the project in VSCode, and nothing more. If preference is given to some Zephyr-related extension, they're free to install and use, as long as the extension is capable of identifying what the environment provides, and uses it as much as possible.

So, this is the motivation, for my case specifically. However, the direct manipulation of `ZEPHYR_SDK_INSTALL_DIR` and other Zephyr-related environment variables in the Terminal breaks many use-cases, even related to other extensions:

[image]

Looking into Zephyr IDE's source code, I see that `ZEPHYR_SDK_INSTALL_DIR` is set to be replaced, at this part of the extension's code:

https://github.com/mylonics/zephyr-ide/blob/20f3f73e6029ce2f388dfb129a8783e5f7135c5c/src/utilities/utils.ts#L231

And `getToolchainDir()` is somewhat hardcoded, and not even in a layout expected for an user of Zephyr SDK (I mean, which references `ZEPHYR_SDK_INSTALL_DIR`... I think?):

https://github.com/mylonics/zephyr-ide/blob/20f3f73e6029ce2f388dfb129a8783e5f7135c5c/src/setup_utilities/workspace-config.ts#L179-L200

So, is there a way to make this behavior optional, or at least configurable to some extent? (Not only for `ZEPHYR_SDK_INSTALL_DIR`, but for `ZEPHYR_BASE` and any other env-var that Zephyr IDE intends to manage)

rijesha, 2025-10-15:
The best way to implement this is to do the following actions:

In the ```reloadEnvironmentVariables``` function the update to "ZEPHYR_SDK_INSTALL_DIR" should be moved into the check for ```if (setupState)```.

Then there are a bunch of checks on wsConfig.activeSetupState in project_utilities and zephyr_utilities. These need to be either removed or a flag should be added to disable the checks if a configuration setting is present in .vscode/settings.json. Another solution that may be better is creating a new dummy setup state called "externally managed" or something similar to keep the logic as is. We would then need a mechanism to select externally managed. This could be a command "use-externally-managed-west-workspace" and also an action button on the west workspace panel.

rijesha, 2025-11-10:
@flisboac can you try prerelease v2.0.10. It should be live tomorrow and see if you have any issues with it.