mylonics/zephyr-ide #190: Irreversible PATH corruption on Windows 11 caused by incorrect Install Step 3 command (Automatic & Manual)

tbitcs, 2025-09-16:
### ⚡ TL;DR

The installation instructions tell users to run

```
setx path "%path%;C:\Program Files\7-Zip"
```

On Windows 11, this command can **irreversibly corrupt PATH** due to `setx` truncation and variable expansion. It wipes or breaks PATH, leaving developer environments unusable. Needs urgent fix in both **Automatic Install** and **Manual Step 3** docs.

---

### Describe the bug

When using **Run Automatic Install**, or when following the **Manual Install Step 3 (Configure Environment)** instructions, the user is told to run:

```
setx path "%path%;C:\Program Files\7-Zip"
```

This command is **dangerous** on Windows 11 and can irreversibly corrupt the system PATH.

---

### Dangers of using `setx path %path%`

* **Truncates your PATH**: `setx` has a 1024-character limit. If PATH exceeds this, everything beyond it is **permanently deleted**, breaking critical applications.
* **Expands environment variables**: `%SystemRoot%` and others get expanded into full paths, destroying their symbolic meaning.
* **Merges user/system PATHs**: User and system PATH entries get flattened, causing duplication and making updates harder.
* **Misleading behavior**: Current session looks fine, but new terminals/apps use the corrupted PATH.

This makes the documented command unsafe and misleading.

---

### Expected behavior

PATH should be preserved and `C:\Program Files\7-Zip` appended safely.

---

### Suggested fix

Replace the bad command with safer alternatives.

**Safest GUI method (recommended):**

1. Search *"environment variables"* in Start Menu → **Edit the system environment variables**.
2. In *System Properties* → click **Environment Variables…**.
3. Under *User variables* or *System variables*, select **Path** → **Edit…**.
4. Click **New** and paste:

   ```
   C:\Program Files\7-Zip
   ```
5. OK to close dialogs.

**PowerShell (safe, no character limit):**

* Append to **User PATH**:

  ```powershell
  $env:Path += ";C:\Program Files\7-Zip"
  ```
* Append to **System PATH**:

  ```powershell
  [System.Environment]::SetEnvironmentVariable(
    'Path',
    [System.Environment]::GetEnvironmentVariable('Path', 'Machine') + ';C:\Program Files\7-Zip',
    'Machine'
  )
  ```

**Command Prompt (cmd.exe)** — safe only if PATH < 1024 chars:

```cmd
setx PATH "%PATH%;C:\Program Files\7-Zip"
```

---

### ⚠️ Proposed Warning Box for Documentation

```markdown
> **⚠️ Warning: Do not use `setx path "%path%..."` on Windows 11**  
> Using `setx` with `%path%` can **irreversibly corrupt your PATH** by truncating it, expanding variables, and flattening user/system entries.  
> - If you must append to PATH, use the GUI Environment Variables editor or the PowerShell examples provided above.  
> - Only use `setx` in `cmd.exe` if you are certain your PATH is shorter than 1024 characters.  
```

---

### Impact

Running the documented command can cause **irreversible corruption of PATH on Windows 11**, requiring manual repair or even a full reset of environment variables. This can leave a developer environment unusable.

---

### Severity / Priority

\:rotating\_light: **Critical** — This issue can irreversibly break Windows environments and should be treated as a top-priority documentation and installer fix.

---

### Labels to apply

* `bug`
* `critical`
* `docs`
* `windows`

---

### Desktop:

* **OS**: Windows 11 v10.0.26100 Build 26100
* **VSCode Version**: 1.104.0
* **Zephyr IDE Version**: 2.0.5
* **Zephyr Version**: (using SDK install path)
* **Zephyr SDK Version**: v0.17.3
* **Terminal**: PowerShell v7.5.2

rijesha, 2025-09-16:
Thanks for reporting. I will address the issue right away.

rijesha, 2025-09-17:
A fix has been pushed. I am going to close this issue as it addresses the setx path.
If there are other issues related to the host tool install it can be made as a new issue.

closed 2025-09-17 by rijesha